PaxInstruments/t400-firmware #125: Implement a SPI flash test

charlespax, 2015-01-05:
Test the SPI flash.

charlespax, 2015-01-08:
Done! The test just checks to see if the initialization is okay.

In the test I'm using https://github.com/PaxInstruments/SPIFlash.

samchoy88, 2015-01-08:
I use this library for SPI flash testing.
https://github.com/LowPowerLab/SPIFlash

It can read/write data and check Chip ID to identify the type. 
easy to use

charlespax, 2015-01-08:
Cool. That's the one I forked.

charlespax, 2015-01-08:
The SPI flash test works fine when the chip is present and working. I desoldered the chip, so I could confirm the test will throw an error when the chip is not working. After some debugging I believe flash.initialize() is not returning a value.

Here is flashTest() in t400-testing.ino.
```
char* flashTest(void){
  u8g.setFont(u8g_font_6x10);
  Serial.print("Flash... ");
  u8g.firstPage();
    do {
      u8g.drawStr( 25, 35, "Flash...");
    } while( u8g.nextPage() );
    
  long nextUpdate = millis() + TIMEOUT;
  Serial.println("debugging");
  Serial.print("  DEBUG: nextUpdate = ");Serial.println(nextUpdate);
  Serial.print("  DEBUG: millis() = ");Serial.println(millis());
  Serial.print("  DEBUG: initialize() = ");Serial.println(flash.initialize());
  while( millis() <= nextUpdate){
    if (flash.initialize()) {           // *** The initialize function does not return anything when
      Serial.println("PASS");              // *** the flash fails to start. Without the chip, this
      return "PASS";                       // *** function just hangs. Must modify the library.
    }
  }
  Serial.print("  DEBUG: millis() = ");Serial.println(millis());
  if (millis() > nextUpdate){
    Serial.println("FAIL... timeout");
    return "FAIL";
  }
}
```

In SPIFlash.ccp the code below does not return a value as is. I can, however, place a `return true;` at the beginning of the function and have the code in t400-testing work properly.
```
/// setup SPI, read device ID etc...
boolean SPIFlash::initialize()
{
  _SPCR = SPCR;
  _SPSR = SPSR;
  pinMode(_slaveSelectPin, OUTPUT);
  unselect();
  wakeup();
  
  if (_jedecID == 0 || readDeviceId() == _jedecID) {
    command(SPIFLASH_STATUSWRITE, true); // Write Status Register
    SPI.transfer(0);                     // Global Unprotect
    unselect();
    return true;
  }
  return false;
}
```

charlespax, 2015-01-08:
This code returns `true` and the test firmware continues as normal.
```
/// setup SPI, read device ID etc...
boolean SPIFlash::initialize()
{
  _SPCR = SPCR;
  _SPSR = SPSR;
  pinMode(_slaveSelectPin, OUTPUT);
  unselect();
  return true;  //  *** This is my debug code
  wakeup();
  
  if (_jedecID == 0 || readDeviceId() == _jedecID) {
    command(SPIFLASH_STATUSWRITE, true); // Write Status Register
    SPI.transfer(0);                     // Global Unprotect
    unselect();
    return true;
  }
  return false;
}
```
This code returns nothing and firmware hangs.
```
/// setup SPI, read device ID etc...
boolean SPIFlash::initialize()
{
  _SPCR = SPCR;
  _SPSR = SPSR;
  pinMode(_slaveSelectPin, OUTPUT);
  unselect();
  wakeup();
  return true;  //  *** This is my debug code
  
  if (_jedecID == 0 || readDeviceId() == _jedecID) {
    command(SPIFLASH_STATUSWRITE, true); // Write Status Register
    SPI.transfer(0);                     // Global Unprotect
    unselect();
    return true;
  }
  return false;
}
```

charlespax, 2015-01-08:
Here is the flash test I've been using.
```
char* flashTest(void){
  SPIFlash flash(FLASH_CS, 0xEF40);  // SPI flash chip
  
  u8g.setFont(u8g_font_6x10);
  Serial.print("Flash... ");
  u8g.firstPage();
    do {
      u8g.drawStr( 25, 35, "Flash...");
    } while( u8g.nextPage() );
    
  long nextUpdate = millis() + TIMEOUT;
  Serial.println("debugging");
  Serial.print("  DEBUG: nextUpdate = ");Serial.println(nextUpdate);
  Serial.print("  DEBUG: AVR = ");Serial.println(AVR);
  Serial.print("  DEBUG: SPI.transfer(0) = ");Serial.println(SPI.transfer(0));
  Serial.print("  DEBUG: 255 & 1 = ");Serial.println(255 & 1);
  Serial.print("  DEBUG: flash.initialize() = ");Serial.println(flash.initialize());
  while( millis() <= nextUpdate){
    Serial.print("  DEBUG: millis() = ");Serial.println(millis());
    if (flash.initialize()) {
    //if (flash.initialize()) {           // *** The initialize function does not return anything when
      Serial.println("PASS");              // *** the flash fails to start. Without the chip, this
      return "PASS";                       // *** function just hangs. Must modify the library.
    }
  }
  Serial.print("  DEBUG: millis() = ");Serial.print(millis());Serial.print("  while loop ended");
  if (millis() > nextUpdate){
    Serial.println("FAIL... timeout");
    return "FAIL";
  }
}
```

charlespax, 2015-01-08:
This works when the chip is installed, but hangs if the chip is not installed. Closing this issue and opening one to address the hanging problem (https://github.com/PaxInstruments/t400-firmware/issues/134).

closed 2015-01-08 by charlespax

LowPowerLab, 2015-01-09:
The issue is with static/noise on the open drain MISO input. That causes a hang in busy() when non 0 statuses are returned. You should add a 10k pulldown or similar weak pulldown on MISO if it's not used to avoid such open drain misreads caused by static or noise in the circuit. This is safe even with the chip present. A time limit check cannot be safely added in the library because some flash chips can take many seconds to finish executing erases or other multi-block or entire chip operations.

charlespax, 2015-01-09:
Reference: https://github.com/LowPowerLab/SPIFlash/issues/4