PaxInstruments/t400-firmware #137: Implement SPI flash timeout

charlespax, 2015-01-09:
On Electronics version 0.9 the SPIFlash library hangs on `flash.initialize()` because it calls `SPI.transfer(0)`, which always returns 255 with IC not present on this hardware version. This is a problem with implementing a SPI flash test in the T400 testing firmware (https://github.com/PaxInstruments/t400-firmware/issues/125). 

There are a few options for working around this.
- We can implement a hardware solution involving pullup or pulldown resistors. We would have to review the SPI devices and make sure a solution is compatible with all of them. See the discussion an http://www.microchip.com/forums/m148739.aspx
- We can modify the Arduino SPI library to have a timeout. We would modify `SPI.transfer(0)`. This would require us to distribute yet another library. Maybe we can get a patch upstream, but that sounds very slow.
- We can implement a version of `flash.initialize()` with a timeout in the SPIFlash library. It might look like `flash.initialize(TIMEOUT)`. If we implement this nicely, we can probably get it pulled upstream.
- We may be able to include something in the t400-firmware that redefines the `initialize()` function or one of the functions between that and the one that hangs, `SPI.transfer(0)`. This would be the best option as it does not require anything outside of our direct implementation.

charlespax, 2015-08-09:
This is a problem I ran into while making test code. Would this be an issue with the shipping firmware?

cibomahto, 2015-08-13:
Moving to icebox, reconsider it if you're making them in 10k quantity.