PaxInstruments/t400-firmware #142: Backlight brightness set with PWM

charlespax, 2015-01-25:
In Electronics version 0.10 we should be putting LCD_BL on a PWM pin, so the brightness can be adjusted. This issues addresses the back end implementation. 

Electronics issue: https://github.com/PaxInstruments/t400-electronics/issues/109

charlespax, 2015-01-25:
This has hardware implications, so I'm moving it to Firmware version 0.2, so we can address it now.

cibomahto, 2015-01-26:
It's on PD7, which is a PWM:

T0/OC.4D/ADC10 (Timer/Counter0 Clock Input or Timer 4 Output Compare D / PWM output or Analog to Digital Converter channel 10)

Timer 4 is being used by the button polling routine at the moment, but it might be possible to rearrange these.

charlespax, 2015-02-04:
This is not essential to the graphing theme. It's a nice-to-have feature. Moving to the UX milestone.