PaxInstruments/t400-firmware #143: Sleep peripherals when idle

cibomahto, 2015-01-26:
The following peripherals can be configured to a low-power mode when not in use:

MCP9800: Sleep mode (current draw lowered from 200uA to 1uA)
W25Q80: Power down mode (current draw lowered from 25uA to 1uA)
DS3231M: None applicable (draws 130mA when inactive)
MCP3424: Enters standby mode automatically when in one-shot mode (down to 1uA)
LCD is TBD (backlight should be off at a minimum)
SD: Cards should draw <200uA when idle, but the software has to be right: http://forum.arduino.cc/index.php?topic=149504.0

There should be an on routine to bring the peripherals up when they are needed, and an off routine to sleep them all before sleeping the processor.

charlespax, 2015-01-26:
Good call. Sounds like this is non-essential, but good for battery life. Moving to BATTERY milestone.

charlespax, 2015-03-19:
Little advantage. Moving to ice box.