PaxInstruments/t400-firmware #145: 60s log interval too fast

cibomahto, 2015-02-12:
30s seems to be fine, but 60s is very fast. Possibly a rollover problem?

charlespax, 2015-02-12:
This is happening with the RTC_INT triggering the readings?

cibomahto, 2015-02-15:
This is with atmega-based timing, not the RTC_INT yet. It'll likely be irrelevant once the RTC triggers the system.

charlespax, 2015-02-19:
We should either move this issue to a future milestone or pull the dependencies into this milestone. @cibomahto what do you recommend?

Dependencies:

- RTC: Set RTC date/time (https://github.com/PaxInstruments/t400-firmware/issues/51)
- RTC: Set interrupt alarms (https://github.com/PaxInstruments/t400-firmware/issues/50)
- RTC: RTC interrupt mask triggers readings (https://github.com/PaxInstruments/t400-firmware/issues/53)

cibomahto, 2015-02-19:
In that case, just drop it- the code with this bug will be rewritten to support RTC timing.

closed 2015-02-19 by cibomahto