PaxInstruments/t400-firmware #147: Graph indicator alignment

charlespax, 2015-02-16:
As of commit 6b4233eadb422f27645603a8306646cb7c448231 the numbers to the right of the graph to match up with the graph lines.
[image]

They appear to be inverted. In the images below you can see the temperature on sensor 1 decreasing while the label "1" moves up.
[image]
[image]
[image]
[image]

charlespax, 2015-02-20:
Adding to GRAPHING

charlespax, 2015-02-20:
Actually, I'm moving it to Firmware version 0.2 because it aligns with the milestone theme of graphing. Also it will need to be fixed to meet the milestone description of, "At the end of this milestone the graphing system should be functional and display correctly."

cibomahto, 2015-02-20:
Fixed in fed283e

closed 2015-02-20 by cibomahto