PaxInstruments/t400-firmware #148: K type temperature range

charlespax, 2015-02-16:
The int32_t tempTypK() function in intypek_constant.h only process temperatures up to 730 C because the variable used is an int. We'll have to change something here to get the full range.

[image]

charlespax, 2015-02-20:
Moving to READINGS.

charlespax, 2015-02-22:
Now that the lookup table is in PROGMEM it appears that we can enable readings above 30382 uV.

All I did was uncomment the rest of the lookup table in typek_constant.h and change `#define TEMP_TYPE_K_LENGTH 101` to `#define TEMP_TYPE_K_LENGTH 165`.

I tested this the best I could, using a light to heat the end of a thermocouple. Here's the graph:
[image]

Note: I looked into this after implementing the code snippet I posted in the Temperatures below 17.6 issue (https://github.com/PaxInstruments/t400-firmware/issues/160). Not sure if one thing affects the other.

cibomahto, 2015-08-15:
The constants are now stored as int32_t, so they fit well within the allotted space. This can be flipped on when some space is freed up in the firmware.

cibomahto, 2015-08-15:
Freed up some space by not polling the RTC for the current time. 8313e34

closed 2015-08-15 by cibomahto