PaxInstruments/t400-firmware #149: Buffer smaller portions of the screen

charlespax, 2015-02-16:
If we are need to free up some ATmega32U4 RAM, we may be able to buffer a smaller amount of LCD content. See https://github.com/PaxInstruments/t400-firmware/issues/81

Adding to ice box.

cibomahto, 2015-02-16:
It's not really buffering any of the screen content- it has a scratch buffer equal to 1/8th of the screen size, which should come to around 128 bytes (128*64/8/8). It's possible to re-use that space when the display update routine isn't running by putting it on the stack, but it's more predictable to leave it as-is. The current split works great for only having to draw the text labels once, since they don't overlap more than 1  display section. Reducing this would probably double the amount of work done on each redraw.

charlespax, 2015-08-09:
Sounds like it's not worth doing and we likely won't need to take any drastic measures. Closing. issue.

closed 2015-08-09 by charlespax