PaxInstruments/t400-firmware #150: LCD refresh rate

charlespax, 2015-02-16:
What can be done to speed up the refresh rate of the LCD? What factors limit the refresh rate?

- We may be able to buffer larger portions of the screen, but that would eat up more MCU RAM (see https://github.com/PaxInstruments/t400-firmware/issues/81)
- Maybe we can speed up the SPI communication
- Maybe it would be help

cibomahto, 2015-02-16:
The correct term here is 'update rate'. The refresh rate is handled by the controller built into the LCD.

The buffer size won't have much of an effect, that's mostly been optimized- see draw() in functions.cpp.

It might be possible to optimize the SPI communication to increase output speed.

charlespax, 2015-02-16:
Here is a relevant post about optimizing the update rate. http://jeelabs.org/2010/11/19/speedier-graphics/

cibomahto, 2015-02-16:
I don't think this is a limiting factor in the system performance.

charlespax, 2015-02-20:
I agree. I was just dumping some ideas. Moving this to the ice box (Formerly "Firmware version :-)").

charlespax, 2015-08-09:
What are the limiting factors?

charlespax, 2015-08-15:
From our conversation I understand that the biggest limiting factor is the processor preparing the buffer to write to the screen. A faster processor with enough ram to buffer the entire screen should be able to update quickly.

This will be important int he future when we move to a better chip. Using the same LCD shouldn't be a problem.

closed 2015-08-15 by charlespax