PaxInstruments/t400-firmware #152: Power button

charlespax, 2015-02-20:
1) When the device is off and the power button is pressed, the device should turn on. This is implemented. The hardware gives power to the device and the software keeps the power on.

2) When the device is in and the power button is pressed for some time interval, the device should turn off.

3) the functionality of 2 should not interfere with the functionality of 1;  after turning on the device the user must release the power button before the device can be turned off.

4) During logging the power button is disabled and the device cannot be turned off via the power button.

charlespax, 2015-02-20:
I'm moving this to UX.

cibomahto, 2015-08-13:
Turns out that the hardware pin is connected to the SS pin (PB0) on Rev12, so the power button can't actually be used. I recommend swapping this with the power on/off output (PB7), as the SS pin can still be used as a GPIO output.

charlespax, 2015-08-13:
For info on the interrupt pins see https://github.com/PaxInstruments/t400-electronics/issues/143

cibomahto, 2015-08-13:
Actually it is best to avoid SS as it gets toggled during SPI init (which then shuts off the board). This works:
Move SW_POWER to PB7 (D11)
Move POWER_ONOFF to PD5 (TXLED/D30)

cibomahto, 2015-08-13:
Implemented in 690b2f6. Requires an update to the bootloader and board definition file as well:
https://github.com/PaxInstruments/ATmega32U4-bootloader/commit/c14d7102692cd88ac141dae1a0a6f0d3ee7ebdd7

Filed an issue in the electronics repo here:
https://github.com/PaxInstruments/t400-electronics/issues/184

closed 2015-08-13 by cibomahto