PaxInstruments/t400-firmware #156: Temperature data persistence

charlespax, 2015-02-22:
In commit ee0c6c6, as the user switches between units (K, C, F), the data gets wiped from the graph. It would be nice if the data persisted while changing units. I imagine this would have some calculation overhead.

Not essential to the minimum viable product. Moving to ice box.

cibomahto, 2015-02-22:
Yeah, that would be nicer. This is actually waiting on automatic graph scaling- once that works, the conversion setup will be similar (all data should be scaled & shifted based on the difference between the previous scale and the new one). The graph doesn't actually know about units. There's an issue open for the automatic scaling already.

charlespax, 2015-02-22:
Should I mark this as a bug and toss it in GRAPHING?

cibomahto, 2015-02-22:
If you want to, sure. I have it marked as a TODO in the code

charlespax, 2015-08-15:
Not necessary to ship, so I'm moving to ice box.

charlespax, 2016-07-30:
It is fine for changing time base to blast the data.

protological, 2016-07-30:
Since the actual temperature values are in the array, it should be pretty easy to do the 'convert' function at display time rather than sample time.  Doing this and removing the reset function call when the button is pressed should allow for  'persistent' data.

This was added to the performance_changes branch

protological, 2016-07-30:
Added in commit d736e22a864211ac280730f84d28d12f07f64866

closed 2016-07-30 by protological