PaxInstruments/t400-firmware #157: Log file name

charlespax, 2015-02-22:
I ran a little test on the log file names. On my SD card I have a bunch of log files named LD0001.CSV through LD0007.CSV. If I were to press log start/stop I would get a file named LD0008.CSV. I took out the SD card, deleted LD0002.CSV and put it back in the T400. I then pressed log start/stop. The T400 made a new log file named LD0002.CSV as expected.

This has me thinking if this is the best behavior. Should the T400 try to make the lowest numbered file name possible or should it make a file named one unit greater than the largest named file on the SD card? In the latter scenario the T400 would have made a file named LD0009.CSV even though LD0002.CSV is missing.

I'm no UX guy, but I imagine users would expect the most recent file to be at the end of a list, just like photos on a camera. I also wouldn't expect people to delete files.

We could store the most recent file name in EEPROM and always increment it. If we replace the LD with digits we can log up to 1,000,000 files before rolling over. We could drop the CSV from the display , keeping it in the actual file name, and make 10,000,000,000 files before rolling over. If we implement user-configurable date/time (https://github.com/PaxInstruments/t400-firmware/issues/153), we can name files as YYYYMMDDhhmmss.CSV and display hhmmss.CSV or "...hhmmss". We may be able to find some more space on the LCD for filename characters too and display the whole filename.

Using firmware in commit 8e7261ded62950411b5c7d05ac078dbe4124a743

Not essential to the minimum viable product. Tossing in ice box.

charlespax, 2015-02-22:
Additionally, I think the first file on a blank SD card should be LD0001.CSV rather than LD0000.CSV. Technically, it's arbitrary. As a user though, it just feels a bit empty to have a file named with all zeros.

charlespax, 2015-08-09:
We should make some decisions about implementing or not implementing this int he MVP.

cibomahto, 2015-08-15:
In 3cf53a7, logging now starts with LD0001.CSV

cibomahto, 2015-08-15:
Cameras usually have a persistent, always-increment mode. Good idea, the only limitation here is lack of storage/ram to implement EEPROM reads/writes. Depends only on further code optimization.

charlespax, 2015-08-15:
I think what we have now i fine. We could have some thing more complex, but I don't think that improves the situation. If I'm recording important data, I will write down the file name. The file names could be randomly generated and the method would still be the same. This is something we can revisit if we ever start writing to the EEPROM. That functionality isn't worth doing just for this.

Now that it starts at LD0001.CVS, I think we can move this to the ice box.

charlespax, 2015-08-15:
This is plenty good. If user testing reveals that people expect a different behavior, we can open another issue. Closing issue.

closed 2015-08-15 by charlespax