PaxInstruments/t400-firmware #158: First reading after log interval change

charlespax, 2015-02-22:
When I change the logging interval all the data is wiped from the graph, which is what we expect (https://github.com/PaxInstruments/t400-firmware/issues/98). The next reading to appear on the graph happens one log interval after the screen is cleared.

This is not a problem if the log interval is one or two seconds, but can be a problem for longer time intervals. I think it would be better to have the first reading take place the moment or within one second of the log interval changing. What do you think?

Using commit 8e7261ded62950411b5c7d05ac078dbe4124a743

charlespax, 2015-02-22:
Also, while the user is waiting for the first reading after the log interval is changed they have to look at al four graph labels sitting at the bottom of the screen.

[image]

cibomahto, 2015-08-13:
The first reading now happens with a second, regardless of mode. There's still some garbage displayed for the first frame, though

charlespax, 2015-08-14:
Should we close this issue and make separate issues addressing the four graph labels and the Y axis labels?

cibomahto, 2015-08-14:
No, they're part of the same issue.

cibomahto, 2015-08-14:
The main thing we can do here is not refresh the display until we take a set of readings.

cibomahto, 2015-08-15:
Implemented in c6c37aa. Adds <=1s delay in responding to the button, but with no visual artifacts.

closed 2015-08-15 by cibomahto