PaxInstruments/t400-firmware #159: Thermocouple disconnect check

charlespax, 2015-02-22:
In functions.cpp at line 236 we have the code:
```
  // Check if the sensor was disconnected
  if(microVolts == 255.99) {
    return OUT_OF_RANGE;
  }
```
It seems like a really specific value to look for. Should the code just check if the value is higher than some number?:
```
  // Check if the sensor was disconnected
  if(microVolts >= 250) {
    return OUT_OF_RANGE;
  }
```

I thought maybe 255.99 was some value that gets passed to GetTypKTemp() to force `return OUR_OF_RANGE`, but I couldn't "255.99" anywhere else in the t400-firmware repository or the Arduino libraries we're using.

Using commit 8e7261ded62950411b5c7d05ac078dbe4124a743

charlespax, 2015-02-22:
In the following I am using commit 8e7261d and have no thermocouples connected.

When I change the function to
```
double GetTypKTemp(double microVolts){
  // Converts the thermocouple µV reading into some usable °C
  Serial.println(microVolts);
  
  // Check if the sensor was disconnected
  if(microVolts == 255.99) {
    Serial.println("OUT_OF_RANGE");
    return OUT_OF_RANGE;
  }...
```
The serial port outputs
```
 0: 2:17, 22.94, -, -, -, -
255992.18
255992.18
255992.18
255992.18
 0: 2:18, 22.88, -, -, -, -
255992.18
255992.18
255992.18
255992.18
```
So, it looks like that `==` check doesn't do anything for disconnected thermocouples.

When I change the function to use `if(microVolts >= 250)`, on the serial port I get
```
 0: 3:33, 22.88, -, -, -, -
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
 0: 3:34, 22.94, -, -, -, -
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
```

If I plug a thermocouple into channel two I get
```
 0: 6: 1, 22.88, -, 23.27, -, -
255992.18
OUT_OF_RANGE
15.63
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
 0: 6: 2, 23.00, -, 23.39, -, -
255992.18
OUT_OF_RANGE
15.63
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
 0: 6: 3, 22.88, -, 23.27, -, -
255992.18
OUT_OF_RANGE
15.63
255992.18
OUT_OF_RANGE
255992.18
OUT_OF_RANGE
```

cibomahto, 2015-08-15:
Addressed in d8e9500 and d8e9500. This magic number was a result of the ADC's output when out of range. I've updated the ADC library to use int32_t instead of double, and to propagate the max value out to make it more explicit. This allowed the check in GetTypKTemp() to be removed.

closed 2015-08-15 by cibomahto