PaxInstruments/t400-firmware #160: Temperatures below 17.6 C

charlespax, 2015-02-22:
Below temperatures of approximately 17.6 C the thermocouples appear as if they are disconnected.

Below is an image of what happened when I put thermocouple channel 2 in a refrigerator. Channel 1 had been in the refrigerator for some time.

[image]

charlespax, 2015-02-22:
Oh I see. This is happening because of
```
  if(microVolts > maxConversion || microVolts < -200){
    return OUT_OF_RANGE;
  }
```

When I comment out that code I am able to measure lower temperatures.

charlespax, 2015-02-22:
Okay, I dug into this a bit more and it seems the minConversion problem is caused from using a double for minConversion and int32_t for the tempTypK table values.
```
  double minConversion = lookupThermocouleData(TEMP_TYPE_K_LENGTH - 73);
  Serial.print("minConversion  = ");Serial.println(minConversion);
```
Outputs to he serial port `minConversion  = 397.00` as one would expect.

```
  double minConversion = lookupThermocouleData(TEMP_TYPE_K_LENGTH - 75);
  Serial.print("minConversion  = ");Serial.println(minConversion);
```
outputs to the serial port `minConversion  = 4294966784.00` when it should output `minConversion  = -392.00`

If I change the minConversion declaration to
```
  int32_t minConversion = lookupThermocouleData(TEMP_TYPE_K_LENGTH - 75);
  Serial.print("minConversion  = ");Serial.println(minConversion);
```
the serial port outputs `minConversion  = -392` as we should expect.

charlespax, 2015-02-22:
Here is code that seems to work. The setup is I have is one thermocouple plugged into sensor port 4 with the other end in my freezer. The code snippit is the change made to the firmware.

```
double GetTypKTemp(double microVolts){
  // Converts the thermocouple µV reading into some usable °C
  
  // Check if the sensor was disconnected
  /*if(microVolts == 255.99) {
    return OUT_OF_RANGE;
  }*/
  
  // Check if it's out of range
  // TODO: Read this once.
  // TODO: Why does minConversion not work?
  int32_t maxConversion = lookupThermocouleData(TEMP_TYPE_K_LENGTH - 1);
  int32_t minConversion = lookupThermocouleData(0);
  //Serial.print("minConversion  = ");Serial.println(minConversion);
  
  if(microVolts > maxConversion || microVolts < minConversion){
    return OUT_OF_RANGE;
  }
  
  double LookedupValue;
  
  // TODO: Binary search here to decrease lookup time
  for(uint16_t i = 0; i<TEMP_TYPE_K_LENGTH; i++){
    int32_t valueLow = lookupThermocouleData(i);
    int32_t valueHigh = lookupThermocouleData(i + 1);
    
    if(microVolts >= valueLow && microVolts <= valueHigh){
      LookedupValue = ((double)-270 + (i)*10) + ((10 *(microVolts - valueLow)) / ((valueHigh - valueLow)));
      break;
    }
  }
  return LookedupValue;
}
```

Serial port output:
```
0: 0:17, 24.06, -, -, -, -16.92
 0: 0:18, 24.06, -, -, -, -16.92
 0: 0:19, 24.13, -, -, -, -16.86
 0: 0:20, 24.06, -, -, -, -16.92
 0: 0:21, 24.06, -, -, -, -16.92
 0: 0:22, 24.13, -, -, -, -16.86
```

cibomahto, 2015-08-15:
Addressed in d8e9500 and d8e9500.

closed 2015-08-15 by cibomahto