PaxInstruments/t400-firmware #165: Arduino boot delay

charlespax, 2015-06-03:
It would be cool if the normal Arduino boot delay is skipped, so we can boot faster. Maybe have the bootloader check for a specific key press combination. If the keys are not pressed, move directly to firmware. If keys are pressed, proceed with the normal delay. This would result in the user needing to hold down keys while flashing new firmware.

charlespax, 2015-06-03:
Maybe this should go in the bootloader repo issues.

cibomahto, 2015-06-05:
The button thing is already implemented, and Caterina does not have a boot delay on regular startup.
[image]

closed 2015-06-05 by cibomahto