PaxInstruments/t400-firmware #169: Sub 1s readings

charlespax, 2015-08-13:
It would be nice to take readings at half seconds. Right now we can't hit that mark without extreme optimization. Sending to the ice box.

cibomahto, 2015-08-15:
It's possible to read and display the data this fast, but writing to SD could possibly take much longer (the write time isn't guaranteed), so this isn't reasonable to do without a fairly major rewrite.

charlespax, 2015-08-15:
I have several MicroSD card sample from several vendors I'm considering. When recording at 1s intervals is the SD write speed a concern or is pretty much everything capable of working at 1s?

cibomahto, 2015-08-15:
Good question, I don't know the answer. My guess is that it's the speed of the arduino that's the limiting factor.

protological, 2016-07-12:
I did some measurements and it looks like we can read 13-14 ADC values per second and the display writing can be done 10 times per second.  I removed unused code and tested each function running as fast as possible in the main loop() function.  If we were to write the SD card once a second with 2 set of data I think we can get to 1/2 second sampling.  This will take a lot of updates but it is possible.

protological, 2016-07-30:
1/2 second sampling was added with a 500ms timer started when the RTC interrupt was received.  When the timer ends another sample is locked in and displayed.

This was added in commit e5cad9540f673d7f2866c56c94ee36aa972c49f9

closed 2016-07-30 by protological