PaxInstruments/t400-firmware #172: Determine charge state

cibomahto, 2015-08-14:
In an effort to get power input/charge state reporting working, I spent some time this evening looking into the behavior of VBAT and BATT_STAT.

I'm charging my battery to test the condition of charged battery now. The circuit looks like it might work, but I'm considering a change to the filter circuit to improve noise immunity for the output.

It turns out it works a little differently than expected:

1. When the device is powered from USB but a battery isn't inserted, the MCP73832 does a battery detect routine and continually turns its output on/off. This causes a square wave at it's pin 1 (stat), which, due to the R1/R17 voltage divider and the large cap C22, ends up showing a voltage of .96V at BATT_STAT. It also shows a voltage of 4.12V at VBAT. This means that the 'usb powered, no battery' state can be determined by measuring BATT_STAT to see if it falls near 1V.

2. When the device is powered from USB and a partially-charged battery is inserted, the MCP73832 seems to do it's job and pulls down it's stat output, making the BATT_STAT voltage .57V. VBAT is then at 4.19V, but this might change as the battery charges. This means that the 'usb powered, charging battery' state can be determined by measuring BATT_STAT to see if it falls near .57V.

3. (testing now) When the device is powered from USB and has a fully charged battery, BATT_STAT should be near 3.3V, and VBAT is probably still something like 4.2V.

4. When the device is powered from Battery (no USB connected), BATT_STAT goes to 3.3V like it is supposed to, and VBAT is 4.06V (though this probably changes with the voltage). This means the 'battery powered, discharging' state can be determined by measuring BATT_STAT to see if it is at 3.3V, but we also need to check for VBUS to disambiguate from state #3.

The decision tree then looks like this:

VUSB not present? DISCHARGING (and measure VBAT_SENSE to estimate the charge level)
VUSB present, BATT_STAT<.7V: CHARGING
VUSB present, .8V<BATT_STAT<1.2V: NO_BATTERY
VUSB present, BATT_STAT>3V: CHARGED

charlespax, 2015-08-14:
BATT_STAT is on A7, so we should be good for measurement.

cibomahto, 2015-08-15:
Implemented in 996fd74dfd. This implementation seems fine. A future suggestion is to remove all the analog components here and read this digitally, using the timer counter mode of the status pin to detect if it's in battery disconnected mode.

closed 2015-08-15 by cibomahto