PaxInstruments/t400-firmware #178: Display shows crazy units when turned on with no probes attached

cibomahto, 2015-08-18:
Should avoid drawing the graph if the labels are out of range

charlespax, 2015-09-02:
In t400.cpp we can change `double temperatures[SENSOR_COUNT];` to `double temperatures[SENSOR_COUNT] = {25,25,25,25};`

Quick and dirty, but it should work. I haven't tested it thought because compiling the current firmware gives `Sketch uses 28,672 bytes (100%) of program storage space. Maximum is 28,672 bytes.`.

charlespax, 2015-09-22:
A crazy graph can be made normal by plugging in a thermocouple and removing it. The graph y-axis will look fine. It will then go crazy again if the units or timebase is changed.

Here is what it looks like.
[image]

charlespax, 2015-10-03:
Fixed in 7d054c3f8ef75802a9b2e6a88c616fdc4baf70a0. "Changed the temperature values in graphReset(), so we don't see crazy values. I'm not sure if I'm doing this the right way. Closes 178. Maybe we should draw the axis after the first reading."

closed 2015-10-03 by charlespax