PaxInstruments/t400-firmware #183: What is the graph cycling implementation plan?

charlespax, 2015-09-22:
This question must be answered before working on #44.

The T400 firmware has reached the limits of the 32u4 flash memory. To implement graph cycling we will need to optimize some code or offload some data to the SPI flash chip.

Answers to these questions would help us make a choice on how we approach this:

- How much additional memory is required to implement #44?
- What are possible areas where we can optimize code?
- How much much memory could we expect to free up through code optimization?
- How much memory can be offloaded to the SPI flash chip?
- How much additional memory is required to implement using the SPI flash chip?

I think a good approach is the following:

- Free up a bunch of memory by truncating the thermocouple lookup table
- Compile the code to determine the available memory, M1
- Use the freed memory to implement the graph cycling feature #44
- Compile the code and determine the available memory, M2
- Now we know we have to free up M1 - M2 more memory. This will help us determine which approach to take.

MLXXXp, 2015-09-22:
Are you using the latest version of avr-gcc? Each new release tends to do a better job of optimization, which reduces code size.

You probably want to compile with the -Os flag to optimize for size.

Also, are you using "link time optimization" using the -flto flag? This can also reduce code size.

https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html

charlespax, 2015-09-22:
Right now we're just using Arduino IDE 1.6.5  which uses gcc version 4.8.1 and calls `avr-g++ -c -g -Os`

charlespax, 2015-10-02:
I removed the junction temperature display #184, which freed up 188 bytes. I'll grab some more by reducing the lookup table size. Then I'll implement the graph cycling feature using existing code as templates, including temperature units button and the code that graphs only probes that are plugged in. Then we'll see where we are.

charlespax, 2015-10-03:
Done.

closed 2015-10-03 by charlespax