PaxInstruments/t400-firmware #202: Graph scaling for F and K

charlespax, 2016-02-15:
"When in Celsius mode, the display properly scales the vertical scale to show all four probes with maximum vertical resolution. However, it doesn't seem to do that on Fahrenheit or Kelvin leaving plenty of unneeded "white" space on the bottom of the screen making the display vertical resolution not nearly as good as the Celsius mode." -Eirc

charlespax, 2016-02-15:
This is in function.cpp.

When the graph is reset the following code is executed.
```
void resetGraph() {
  graphCurrentPoint = 0;
  graphPoints = 0;
  
  graphMin = 25;  // TODO: sliding scale?
  graphScale = 1; // in 10ths

  maxTemp = 29;
  minTemp = 25;
}
```
This will make lower bound always be 25, but not necessarily 25 C. This is just 25 of whatever unit we have. If we then switch to Kelvin or Fahrenheit the lowest point on the graph will still be 25 K or 25 F respectively.

I can probably figure out a way around this. Ultimately thought, implementing a sliding temperature axis based on the total on-screen temperature range will solve this and be the proper implementation.

MLXXXp, 2016-02-15:
It would also be nice if the user could manually set the upper and lower bounds, to lock the range displayed. The probe numbers on the right of the screen could show a "+"  beside them and be pinned at the top, or a "-" and be pinned at the bottom, for temperatures out of range. Or alternatively, instead of "+" or "-" used arrows or triangle pointers "▲" or "▼".

It would be desirable to indicate that you are in fixed range mode. Perhaps change the graph "tick", to the right of the upper and lower bounds values, to a triangle pointer "►".

Deciding on the buttons used to control this would be a challenge. Perhaps make it work like many digital watches and timers do:

1. Press and **hold** the *thermometer* button until the lower range number starts flashing, then release it.
- Use the *record* button to increase the lower value and the *time* button to decrease it. The value continues to flash while being adjusted.
- Press the *thermometer* button to lock the lower bound. The lower bound number goes solid and the  the upper range number starts flashing.
- Use *record* and *time* to adjust the upper value while it's flashing.
- Press *thermometer* a third time to lock the upper bound and return to normal operation, with a fixed scale.

It would be nice to have the values auto increment/decrement if *record*/*time* were held down.

To return to auto scale mode:

1. Press and hold the *thermometer* button to enter range select mode, as above.
2. While either range number is flashing, Press and **hold** *thermometer* a second time. This will return to normal operation in auto scale mode.

If you wanted to support selecting thermocouple types, as suggested in issue #200, you could have pressing the *graph* button switch types when either range value is flashing.

charlespax, 2016-07-13:
@protological can we close this now that https://github.com/PaxInstruments/t400-firmware/pull/219 is merged into master?

protological, 2016-07-14:
Yes, scaling works

closed 2016-07-14 by protological