PaxInstruments/t400-firmware #212: Remove all floating point operations

charlespax, 2016-04-12:
There are many bast ten floating point operations. If possible, let's make things base 16 and use bit shift math. This should save a bunch of flash space since the floating point library is pretty big.

protological, 2016-07-11:
I've looked at the t400 code and I think the floating point operations can be replaced with int16 values.  The int16 would have a range of -35535 to 35535, if the values were stored as base 10 with a resolution of 1/10th degree the supported range would be -3553.5C to 3553.5C.  The base 10 division and modulo operations would be must faster and take less code space than the floating point.

protological, 2016-07-12:
I think I need some more information on the temperature vs microvolt lookup in the code.  I see the following functions, where did they come from? If we can eliminate some of these floating point operations the code will be smaller and faster

From GetJunctionVoltage():
    i = jTemp/10 + 27;
    jVoltage = valueLow - TK_OFFSET + (jTemp - (i*10-270)) * (valueHigh - valueLow)/10;

From GetTypKTemp():
    LookedupValue = ((float)-270 + (i)*10) + ((10 *(float)(microVolts - valueLow)) / ((float)(valueHigh - valueLow)));

charlespax, 2016-07-13:
@protological  Can this be closed now that https://github.com/PaxInstruments/t400-firmware/pull/220 is merged?

protological, 2016-07-14:
Not all the float operations have been removed.  The mentioned algorithms use floating point values.  Maybe we need to keep some floats

protological, 2016-07-30:
I've managed to reduce the floating point math quite a bit, however there are still a few hold outs

t400.ino
void readTemperatures(): lines 271-284

functions.c
int32_t celcius_to_microvolts(float celcius): Argument is float and then converted to uint8_t
int16_t microvolts_to_celcius(int32_t microVolts): line 517, 525

If we can get some details on the math in these functions I might be able to translate them to integer math operations