PaxInstruments/t400-firmware #221: Y-axis location based on number of digits

charlespax, 2016-07-13:
When the temperature has two digits the graph y-axis looks like this.
[image]

In commit 43543b7205458c8e6638093725c884eada44a73c the y-axis will shift to the right when the number of digits increases from two to three like this.
[image]

In the most recent commit 9fe18bb790af0158d80564159aebaab4549c275b the y-axis does not shift to the right. It looks like this.
[image]

This is probably happening around line 155 in functions.cpp
```
   // Calculate the number of axes digits to display
 -  if(graphMin + graphScale*4 > 999 || graphMin < -99) {		 +  axisDigits = 2;
 +  if((min + (graphScale*4)) > 9999 || min < -999) {
      axisDigits = 4;		      axisDigits = 4;
    }		    }
 -  else if(graphMin + graphScale*4 > 99 || graphMin < -9) {		 +  else if((min + (graphScale*4)) > 999 || min < -99) {
      axisDigits = 3;		      axisDigits = 3;
    }		    }
 -  else {		 +
 -    axisDigits = 2;		 +  return;
```

charlespax, 2016-07-13:
Additional information:

If we start with two digit temperatures like this...
[image]
and heat things up, we get wonky graphics.
[image]

But if we heat the thermocouple before turning on the T400, we can start with only three digit numbers. Then we get a nice graph like this.
[image]
We can keep our nice graph even if we let things cool down.
[image]

Also notice that in the original firmware the two digits are right justified while on the newer firmware they are left justified. They should be right justified.
[image]

charlespax, 2016-07-13:
Also weird with negative numbers.
[image]

In that image you may have also noticed the [double negative](https://github.com/PaxInstruments/t400-firmware/issues/222).

protological, 2016-07-14:
The logic for the number of character spaces needs to be updated for the new 1/10th temperature values.  I have some updated code and will push a fixed branch

protological, 2016-07-30:
Fixed in commit 78bd24936d339fcaf88add8d64a5a1af9bf23ccc

closed 2016-07-30 by protological