PaxInstruments/t400-firmware #225: Random noise data points

protological, 2016-07-14:
While working on updated graphics functions I noticed that randomly, individual pixels would be enabled on the screen when I wasn't writing them. When the next update of the screen is done the pixels are off.

I checked out commit 43543b7205458c8e6638093725c884eada44a73c and the random pixels were found there too. (This is before modifications) I added a quick test function to fake temperature data in a sin wave and was able to catch a few on the screen.  I've looked at the modified code and I don't think these pixels are due to programming, there might be an issue with the u8g library.

[image]

Thoughts?

charlespax, 2016-07-15:
I've also noticed this happen sometimes. I did not find anything when searching for "[u8glib random pixels](https://www.google.com/search?q=u8glib+random+pixels&oq=u8glib+random+pixels)."

The u8glib driver used for this LCD controller chip may not be configured correctly. u8glib did not support it natively, so I found one of their LCD drivers that basically worked and modified it until it did the job. There is a good chance I did something incorrectly.

protological, 2016-08-01:
Looking in the u8glib folder it appears the only pax files are a class definition for the LCD.  Were files in the utility/ or examples/ folders modified?  I can dig into the modified code and see if there is something that jumps out